vmatare/thinkfan #247: Fan's keep running when putting laptop into sleep/suspend mode ThinkPad T14 Gen 4 i7-1335U

bremme, 2024-08-24:
I have a Lenovo ThinkPad T14 Gen 4 with a i7-1335U running Ubuntu 24.04 and I've compiled thinkfan from source. The tool seems to work fine, except for the fact that when I put the laptop into sleep the fans don't stop running. This makes sleep pretty unusable as you can imagine.

Maybe important to add this system only supports s2idle or S0 sleep. I've tried many things but could not get it to work properly. After much trial and error I found a workaround:

```shell
sudo systemctl edit thinkfan-sleep
# add the following line

[Service]
ExecStart=sh -c "echo 'disable' > /proc/acpi/ibm/fan"
```

I was a bit worried that this is potentially unsafe, but if I just watch ` /proc/acpi/ibm/fan` I see the fan's will be enabled automatically after waking from suspend so for me this workable I think.

But I was wondering why this happens in the first place and if there is no other proper way to fix it perhaps? In any case this workaround might be useful for other people struggling with the same issue.

IvanSafonov, 2024-09-28:
I have the same laptop and the same problem. And it's thinkpad specific issue. It happens when you set manual levels like `level 1`. Bios for some reason does not turn it of afterwords. The only solution I have found is the same, change level before sleep. And I think it's safer to set `level auto`. There is a small chance that laptop won't enter sleep mode and the fan will be disabled for some time. The other problem is that now you have two concurrent level changes. Yesterday I'he pushed a new version of my own fan control service, and I think I [solved](https://github.com/IvanSafonov/fanctl/blob/main/conf/fanctl.yaml#L46-L48) this issue.

jolars, 2024-11-07:
I believe this is a duplicate of https://github.com/vmatare/thinkfan/issues/181 I believe.